jeelabs/jeelib #78: "Serial" Errors when using recommended attiny core

stahlfabrik, 2015-06-22:
I need help with the jeelib. It compiles fine with the rather old (abandoned?) attiny core from google code: https://code.google.com/p/arduino-tiny/

But nowadays Arduino recommends this core to be used with the Attinys:
https://github.com/damellis/attiny

This core has very nice integration for thew new Arduino-IDE-Board-Manager. Sadly, jeelib cannot be used with this core. The error generated  from the most basic sketch (used with Attiny84 @8 Mhz) is:

In file included from /Users/user/Documents/Arduino/libraries/jeelib-master/JeeLib.h:17:0,
                 from sketch_jun22a.ino:1:
/Users/user/Documents/Arduino/libraries/jeelib-master/Ports.h:713:49: error: 'Serial' was not declared in this scope
     InputParser (byte size, Commands*, Stream& =Serial);
                                                 ^
/Users/user/Documents/Arduino/libraries/jeelib-master/Ports.h:714:60: error: 'Serial' was not declared in this scope
     InputParser (byte* buf, byte size, Commands*, Stream& =Serial);

How can I fix the compilation problems?

JohnOH, 2015-06-22:
I use https://github.com/jcw/ide-hardware
which I think is the Jeelabs approach the ATTiny range.

stahlfabrik, 2015-06-22:
Thank you for the link. The core that you link to is basically the arduino-tiny core from google code repository with a few (little) modifications for the Jeenode, right?

As I have TinyTx Boards that I would like to develop for, I would like to stick to the more general cores. 

This page introduced my to the damellis core as the recommended core.

http://playground.arduino.cc/Main/ArduinoOnOtherAtmelChips

As far as I understand, the problem is that the google-code-core comes with an included Serial-Software-Library that the jeelib makes use of. 
I want to ask if the serial parts are absolutely necessary for jeelib to work or if it would be possible to not include the Serial code when running on Attiny? That would be very nice!

jcw, 2015-06-22:
Ah, ATtiny support has progressed since when I had to solve all that!

I'll be happy to accept patches (and help, where possible) to make JeeLib compatible with the approach you describe.

The console code in the google-code design could probably be dropped, or disabled somehow.

stahlfabrik, 2015-06-22:
To understand the situation a little better: Do you know (remember;-)) why the rf12demo.ino uses the "bit banging" code for the serial input when targeted to the tinys?

Is the ready to use "SoftwareSerial" Library worse? Does something speak against it? I try to figure out if that Library (comes with Arduino) is not the way to go with Tinys nowadays.

The problem is: I am new to Arduino/Atmel/all that stuff - I am thus hoping on help from people who know this stuff and the existing library landscape.

JohnOH, 2015-06-22:
The bit banging can't be avoided in one form or another since the Tiny84, our primary board in the Tiny family doesn't have a UART. SoftwareSerial was too large when alongside RF12Demo so I cobbled together what you see today..

damellis, 2015-06-25:
I've been maintaining the ATtiny core and wanted to add some more information. The problem seems to be that jeelib assumes (in Ports.h) that if Stream_h is defined, that Serial will also be defined. That is, the InputParser class is compiled if Stream_h is defined, but it uses "Serial" as a default argument value (lines 713 and 714), even if that object isn't defined. My ATtiny files use the default Arduino core, which includes Stream.h but there aren't any hardware serial ports on the ATtiny's I'm supporting. The SoftwareSerial library works, though, and uses Stream.

Is it possible to change the library so it can deal with situations in which Stream is defined but Serial isn't?

anroots, 2015-08-08:
I bumped into this one too. Installed Attiny boards using the new Boards Manager of Arduino 1.6, installed Jeelib, tried to compile a sketch that used `Sleepy::loseSomeTime`, got an error about `Serial` not being defined.

It would be really nice it Jeelib worked on the default core that comes from Arduino. This would lower the entry barrier and allow more 'simple' people like me (who don't necessarily understand how to hack libraries or what goes on behind the scenes) use the library.

My current workaround is to drop Jeelib from the project (with a heavy heart - I tried the Attiny conf from Google Code with an older version of Jeelib (feb 2014), it did not work) and go with Sparkfun [Low Power Attiny](https://learn.sparkfun.com/tutorials/h2ohno/low-power-attiny) (code [here](https://github.com/sparkfun/H2OhNo/blob/master/firmware/WatchDogTest/WatchDogTest.ino)). It's messier than `Sleepy` and I don't understand everything that goes on, but I got the power consumption down with this.

Guys who know more of C / Jeelib - fingers crossed for a pull request that makes Jeelib work with default Attiny85! :)

JohnOH, 2015-08-08:
@anroots, which aspect of jeelib are you using with your Tiny85? Could you post your code?

JohnOH, 2015-08-08:
I can't reproduce this Sleepy problem with my Arduino 1.6.0. Boards manager came in 1.6.3 I think, I will try a later Arduino version.

JohnOH, 2015-08-08:
I have installed Arduino 1.6.5.r2 and have still not seen a problem using Sleepy. Can you post a link to how you added the Tiny boards using Boards Manager. I am still using the Jeelabs Tiny libraries as per: https://github.com/jcw/ide-hardware/

anroots, 2015-08-08:
@JohnOH, you're right, the main functionality I'm interested in is Sleepy (making my code energy-efficient, yay! :)).

An example would be the sketch at [anroots/attiny85-melody-module](https://github.com/anroots/attiny85-melody-module). It tries to use Sleepy, but fails. When I developed  it two years ago with an older Arduino IDE and Jeelib, the error was not present. With Arduino IDE 1.6.5 and Attiny by David. A Mellis v 1.0.1 installed via Arduino Boards Manager (Attiny 85, 8 MHz internal) I'm unable to compile:

```
Build options changed, rebuilding all
In file included from /home/ando/Arduino/libraries/jeelib/JeeLib.h:17:0,
                 from melody_firmware.ino:15:
/home/ando/Arduino/libraries/jeelib/Ports.h:713:49: error: 'Serial' was not declared in this scope
     InputParser (byte size, Commands*, Stream& =Serial);
                                                 ^
/home/ando/Arduino/libraries/jeelib/Ports.h:714:60: error: 'Serial' was not declared in this scope
     InputParser (byte* buf, byte size, Commands*, Stream& =Serial);
                                                            ^
Error compiling.
```

I'm actually using an older version of JeeLib here, but the error is the same in `master`.

JohnOH, 2015-08-08:
I don't have the Tiny85 defined but it build fine for the Tiny84. I am using the jeelabs core as mentioned in my earlier post:
Build options changed, rebuilding all
Using library jeelib in folder: C:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib (legacy)

C:\Users\John\Documents\arduino-1.6.5-r2\hardware\tools\avr\bin\avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=attiny84 -DF_CPU=8000000L -DARDUINO=10605 -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\cores\tiny -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\variants\standard -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib C:\Users\John\AppData\Local\Temp\build8254158818092276568.tmp\sketch_aug08a.cpp -o C:\Users\John\AppData\Local\Temp\build8254158818092276568.tmp\sketch_aug08a.cpp.o 
C:\Users\John\Documents\arduino-1.6.5-r2\hardware\tools\avr\bin\avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=attiny84 -DF_CPU=8000000L -DARDUINO=10605 -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\cores\tiny -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\variants\standard -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\utility C:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\Ports.cpp -o C:\Users\John\AppData\Local\Temp\build8254158818092276568.tmp\jeelib\Ports.cpp.o 
C:\Users\John\Documents\arduino-1.6.5-r2\hardware\tools\avr\bin\avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=attiny84 -DF_CPU=8000000L -DARDUINO=10605 -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\cores\tiny -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\variants\standard -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\utility C:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\PortsBMP085.cpp -o C:\Users\John\AppData\Local\Temp\build8254158818092276568.tmp\jeelib\PortsBMP085.cpp.o 
C:\Users\John\Documents\arduino-1.6.5-r2\hardware\tools\avr\bin\avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=attiny84 -DF_CPU=8000000L -DARDUINO=10605 -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\cores\tiny -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\variants\standard -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\utility C:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\PortsLCD.cpp -o C:\Users\John\AppData\Local\Temp\build8254158818092276568.tmp\jeelib\PortsLCD.cpp.o 
C:\Users\John\Documents\arduino-1.6.5-r2\hardware\tools\avr\bin\avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=attiny84 -DF_CPU=8000000L -DARDUINO=10605 -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\cores\tiny -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\variants\standard -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\utility C:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\PortsRF12.cpp -o C:\Users\John\AppData\Local\Temp\build8254158818092276568.tmp\jeelib\PortsRF12.cpp.o 
C:\Users\John\Documents\arduino-1.6.5-r2\hardware\tools\avr\bin\avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=attiny84 -DF_CPU=8000000L -DARDUINO=10605 -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\cores\tiny -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\variants\standard -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\utility C:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\PortsSHT11.cpp -o C:\Users\John\AppData\Local\Temp\build8254158818092276568.tmp\jeelib\PortsSHT11.cpp.o 
C:\Users\John\Documents\arduino-1.6.5-r2\hardware\tools\avr\bin\avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=attiny84 -DF_CPU=8000000L -DARDUINO=10605 -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\cores\tiny -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\variants\standard -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\utility C:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\RF12.cpp -o C:\Users\John\AppData\Local\Temp\build8254158818092276568.tmp\jeelib\RF12.cpp.o 
C:\Users\John\Documents\arduino-1.6.5-r2\hardware\tools\avr\bin\avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=attiny84 -DF_CPU=8000000L -DARDUINO=10605 -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\cores\tiny -IC:\Users\John\Documents\arduino-1.5.4r2\Tiny\hardware\jeelabs\avr\variants\standard -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib -IC:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\utility C:\Users\John\Documents\arduino-1.6.5-r2\libraries\jeelib\RF12sio.cpp -o C:\Users\John\AppData\Local\Temp\build8254158818092276568.tmp\jeelib\RF12sio.cpp.o 
[... cut ...]
C:\Users\John\Documents\arduino-1.6.5-r2\hardware\tools\avr\bin\avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 C:\Users\John\AppData\Local\Temp\build8254158818092276568.tmp/sketch_aug08a.cpp.elf C:\Users\John\AppData\Local\Temp\build8254158818092276568.tmp/sketch_aug08a.cpp.eep 
C:\Users\John\Documents\arduino-1.6.5-r2\hardware\tools\avr\bin\avr-objcopy -O ihex -R .eeprom C:\Users\John\AppData\Local\Temp\build8254158818092276568.tmp/sketch_aug08a.cpp.elf C:\Users\John\AppData\Local\Temp\build8254158818092276568.tmp/sketch_aug08a.cpp.hex 

Sketch uses 3,338 bytes (40%) of program storage space. Maximum is 8,192 bytes.

JohnOH, 2015-08-08:
Could you take a look at the README for https://github.com/jcw/ide-hardware/

Perhaps we need to add the Tiny85 or adopt the new Boards Manager.Sleepy is only a few lines of code.

JohnOH, 2015-08-08:
Place holder: http://highlowtech.org/?p=1695

anroots, 2015-08-09:
I've installed `jcq/ide-hardware` and chosen JeeNode Micro from the boards menu. I'm unable to compile even the most basic example (Examples -> Analog Read).

```
In file included from /home/ando/Arduino/hardware/jeelab/avr/cores/tiny/WConstants.h:1:0,
                 from /home/ando/Arduino/hardware/jeelab/avr/cores/tiny/WInterrupts.c:36:
/data/bin/arduino-1.6.5/hardware/tools/avr/avr/include/math.h:426:15: error: expected identifier or '(' before 'double'
 extern double round (double __x) __ATTR_CONST__;
               ^
/home/ando/Arduino/hardware/jeelab/avr/cores/tiny/wiring.h:136:26: error: expected ')' before '>=' token
 #define round(x)     ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
                          ^
Error compiling.
```

I think this might be because I'm using the latest Arduino & Linux. My specs:

* Ubuntu 15.04
* Arduino 1.6.5
* Jeelib hardware from `master` as of 2015-08-09

To be fair, jcw/ide-hardware does mention that it's meant to be used with Arduino 1.5.x series.

Going back to the topic of the original issue - @damellis summarises the solution to @stahlfabrik (and mine) problem about Jeelib requiring `Serial` on Attiny8* devices:

> Is it possible to change the library so it can deal with situations in which Stream is defined but Serial isn't?

JohnOH, 2015-08-09:
I have added some conditional statements around the problem area - I wonder if that fixes it. The test branch is https://github.com/jcw/jeelib/tree/SerialErrors

stahlfabrik, 2015-08-09:
Hi JohnOH,

I am happy to see that you start to implement an workaround for this! Thanks!

With master, I get this error:
In file included from /Users/xx/Documents/Arduino/libraries/jeelib/JeeLib.h:17:0,
                 from serialReceiver.ino:2:
/Users/xx/Documents/Arduino/libraries/jeelib/Ports.h:713:49: error: 'Serial' was not declared in this scope
     InputParser (byte size, Commands*, Stream& =Serial);
                                                 ^
/Users/xx/Documents/Arduino/libraries/jeelib/Ports.h:714:60: error: 'Serial' was not declared in this scope
     InputParser (byte* buf, byte size, Commands*, Stream& =Serial);
                               

##########

With the current SerialErrors branch, I get this error:

/Users/xx/Documents/Arduino/libraries/jeelib/RF12.cpp: In function 'void rf12_configDump()':
/Users/xx/Documents/Arduino/libraries/jeelib/RF12.cpp:749:5: error: 'Serial' was not declared in this scope
     Serial.print(' ');

So there should be one (or more?!) code locations in jeelib that use the Serial Object, that is not available on Atiny (Atiny84 for me)

I use Sleepy and RF12M...

stahlfabrik, 2015-08-09:
Maybe rf12_configDump() could be rewritten to take an SoftwareSerial Object as an argument?
Or maybe it could return a long string, to be dumped by the caller?

The next problematic function is: RF12::poll()

There are many if (DEBUG) statements. Maybe they could be also surrounded by #ifdef Serial...

When I temporarily take away the whole function, compilation goes through. So maybe those are all places that need to be adopted to the Attiny Core.

I am totally not sure if it would be possible to rewrite both functions in a way that it could just use an existing SoftwareSerial object. My current skills end there:-)

JohnOH, 2015-08-10:
I am on the wrong track here.  Jeelib requires a serial output capability. For the  Tiny84 hardware this is supplied by the Tiny core. It appears that the latest Tiny cores have removed support for serial. The expectation presumably being the  inclusion of the full output & input capability of software serial. 

To enable use if the latest Tiny core, rather than excluding references to serial we need to provide serial support, via the core or in some other way.

JohnOH, 2015-08-10:
Previously the code size of the full SoftwareSerial was a problem hence the use of the core feature providing serial output only.

JohnOH, 2015-08-10:
I believe that the code from http://www.ernstc.dk/arduino/tinycom.html was provided by the core the supports the current jeelib.

stahlfabrik, 2015-08-10:
Is there a real need for serial in jeelib? The rf12_configDump could be replaced by e.g. a series of getters (e.g. getNodeId).
For the RFM12 to work, no serial is needed.

The debug output could be removed completely.

For sleepy - no serial activity is required.

Maybe the remaining parts that DO need serial could be rewritten to support SoftwareSerial or Serial - whatever the user wants to use - and the hardware supports.

Best regards

JohnOH, 2015-08-10:
I see your point but jeelib supports the Jeelabs products and Serial support on the T84 platform is required. It appears that this Serial support (TinyDebugSerial) has been withdrawn from the Tiny core you are currently using.

I have re-opened your issue on @damellis https://github.com/damellis/attiny/issues/70 to investigate bringing back TinyDebugSerial into his core.

thetrung, 2019-10-05:
Wow.. now I'm having this issue too. Anyone fixed ?

JohnOH, 2019-10-05:
Its been a while, you could take a look at https://github.com/SpenceKonde/ATTinyCore

thetrung, 2019-10-05:
Oh... I'm still using ATTiny from Board Manager Libraries... Also, I hide all Serials temoprarily in this Library when they encounter ATTiny45/85. Thanks for the suggestion !

closed 2019-10-15 by JohnOH