jeelabs/jeelib #87: RF12 support for new features in SPI library

FernandoGarcia, 2015-10-02:
Hi!

I would like to know if you will change the RF12 library to support the new features of SPi library as explained here: http://www.dorkbotpdx.org/blog/paul/spi_transactions_in_arduino
This new features seems solve many problem of conflict between SPI devices.

Best regards.

jcw, 2015-10-02:
Sounds like a good idea. I know Paul Stoffregen looked into this for the RF12 library quite a while ago, but I wasn't sure at the time whether it'd be stable and ready for prime time. Looks like this would be an excellent improvement. I'd welcome a pull request to add this, if anyone gets to it before I can look into it.

FernandoGarcia, 2015-10-02:
I would like to help but I'm not good to work with core. :(