jeelabs/jeelib #88: RFM69 Short packet receipt hang

JohnOH, 2015-10-29:
I think there is a problem with short non jeenode packets passing into the driver. When rf12_len contains a large value but the actual packet length is short, around 5 bytes then TXRECV doesn't appear to exit. This is a placeholder don't do anything on it since my testing and workaround is with RFxConsole. I think master branch will also be suffering.

JohnOH, 2016-04-11:
case TXRECV:
        if (rxfill >= rf12_len + 5 || rxfill >= RF_MAX) {

The above line in RF69::recvDone_compat, part of RF69.cpp I think has a problem. If a noise packet arrives and places a valid length value in rf12_len but doesn't provide sufficient bytes into the FIFO to satisfy the above condition then the RF69::recvDone_compat code will loop indefinitely.

jcw, 2018-09-13:
I'll leave this open, but am un-assigning myself from this issue.
Won't be touching this code any day soon, not even sure it's still relevant.
Feel free to close or elaborate if needed.

closed 2018-09-13 by JohnOH