jeelabs/jeelib #100: encryption for RF69_COMPAT

Ant1Zykl0n, 2016-12-27:
rf12_encrypt(KEY) is working fine when using rf12 modules (without RF69_COMPAT). When using rfm69cw modules with RF69_COMPAT sending and receiving is working fine but the encryption is just ignored (seems to be not implemented). Any ideas?

jcw, 2016-12-27:
Ah, that's probably a bug - with the RF12 driver, an XTEA-based software implementation is used, whereas the RF69 driver uses built-in h/w AES for native mode. I suspect that the compat mode has never been used or implemented.

Ant1Zykl0n, 2016-12-27:
Is it possible to use this library in native mode - without the RF69_COMPAT flag (when every module in my network is using the rfm69cw) and enable encryption? As I cannot find any examples for that.

jcw, 2016-12-27:
Whoops, had a better look - you're right, the Arduino/JeeLib version of RF69 does not support encryption in native mode either.

I can not fix this right now - if you're willing to wait until end of next week, I can have a go at either bringing the RF12 encryption logic into the RF69 compat mode driver, or enabling the AES mode in the native version (I'm not sure at this point which will be easier).

Ant1Zykl0n, 2016-12-27:
Sure :-) Thank you.

FullyScaled, 2017-05-22:
Currently having the same problem. Any chance you will invest time into this in the near future?

jcw, 2017-05-22:
I'm afraid that won't happen any day soon (at least not by me), despite my earlier promise. It's been quite a while since I worked on the RF12 and RF69 compatibility code in JeeLib.

JohnOH, 2017-05-22:
It might be interesting to remove line 32 from RF69_compat.h in jeelib. 
`#define rf12_encrypt        rf69_encrypt`
Not tested it and no guarantees!

JohnOH, 2017-05-22:
On second thoughts that would work on the wrong buffer and thereby not achieve the objective.